cy384/ppk_usb #1: Possible Updates?

danielfrausto, 2015-04-12:
hi i have been following your work. Might it be possible for you to help port this code to leonardo? 
Also i am using landware gotype keyboard for pam III? can you help?

cy384, 2015-04-12:
I'm fairly confident the code should run on a Leonardo with no changes (or just changing some of the pin numbers, maybe).

As for the landware keyboard, unfortunately, I can't find any technical information about it online (and I don't have one), so it would be harder to figure out.  You would probably need a logical analyzer or something to capture and reverse engineer the communications.

DB35M, 2015-04-13:
Hi i  am also using leonardo. i am having problems with pin 15 from your micro pro... there is no pin 15 on leonardo. I have read Micro pro schematic and pin 15 i s SCK but on leonardo that pin goes to ICP.

cy384, 2015-04-15:
You should be able to change that (the PULLDOWN_PIN macro) to basically any other pin you want to wire up, or just ignore it and pull down using GND.

closed 2015-05-03 by cy384