cy384/ppk_usb #2: Any way to debug?

hryamzik, 2015-10-29:
Hi @cy384!

I've connected everything exactly as you've described and it doesn't work. Should the board always appear in the system as "Leonardo" or should there be a different name in a keyboard mode?

cy384, 2015-10-29:
Hi, just added a debug option, try it out by grabbing the latest code and recompiling with DEBUG set to 1.  It should at least give you an idea of where in the process it's failing to work (it will print to the arduino serial monitor).  The board will continue to show up as a leonardo in lsusb or similar lists.

Beyond looking at that, checking your wiring again, make sure you defined PPK_VERSION correctly for your keyboard, maybe poke at it with a multimeter/logic analyzer if you have one, dig into the code and the specification for the keyboard, and hope you don't have any dead hardware.

If you have an extra atmega32u4 based arduino, you could try wiring it up on a solderless breadboard and use test hooks to connect it to the keyboard until you can get it working (which is how I initially did mine before the 3d printed parts).

closed 2015-10-29 by cy384

hryamzik, 2015-10-29:
Got it, thanks!

hryamzik, 2015-10-29:
@cy384 the bad news is that I can see nothing in the serial console. The good news is that it actually started working, should have been a contact issue on a bread board.

Let me ask two more questions:

1. What's the reason for wiring pins 6 and 8 (in a palm V version)?
2. Can I just replace pin 15 with pin 9 (I would like to use an SMD resistor)

cy384, 2015-10-29:
Regarding the issues with the serial console, often I have to make sure manually that it's choosing the right usb/serial port every time I connect the device.

1.  The two reasons for the strange wiring: firstly, simply getting everything to physically fit nicely, and secondly, for reasons unknown to me I could only get the keyboard serial connection to work on certain pins (which I found by just testing them one by one).  I suspect the latter has something to do with interrupts used by the software serial library, but didn't investigate further once I found a way to make it work.

2.  You could replace 15 with any unused I/O pin if you adjust PULLDOWN_PIN, or even just use ground.  It's only used to pull down the serial pin.  15 was just located conveniently for a normal 1/4W through-hole resistor.