cy384/ppk_usb #4: Error uploading

alchle, 2016-01-05:
cy384,

Found this great post on github and am trying it out.  I 3D printed the parts (fits nicely) and am ready to upload the code.  I want to test it on an Arduino Uno first before purchasing a Leonardo and maybe that is my problem. I was hoping that I could connect the Arduino Uno to my tablet and see how well it works, it has a USB port and built in serial communication like the Leonardo.

When I upload the code (changed to Palm V) I get the following errors... looks like it doesn't like the config keymap functions.  Board in Arduino IDE is Arduino Uno, and programmer is USBtinyISP (I never change this setting)

Palm_Keyboard.ino: In function 'void config_keymap()':
Palm_Keyboard:87: error: 'KEY_LEFT_GUI' was not declared in this scope
Palm_Keyboard:107: error: 'KEY_CAPS_LOCK' was not declared in this scope
Palm_Keyboard:108: error: 'KEY_TAB' was not declared in this scope
Palm_Keyboard:109: error: 'KEY_LEFT_CTRL' was not declared in this scope
Palm_Keyboard:120: error: 'KEY_LEFT_ALT' was not declared in this scope
Palm_Keyboard:139: error: 'KEY_BACKSPACE' was not declared in this scope
etc... 
for all key_map in both functions:  'void config_keymap()' & void config_fnkeymap()' 

Please help.

Thanks

cy384, 2016-01-05:
The Arduino Uno isn't able to easily act as a USB keyboard, so it doesn't have the KEY_LEFT_GUI etc. available for this code.  If you want to just see if you can get the serial stuff working, you could try experimenting with this quickly edited version of the code, which will just print the keypresses to the serial console: https://gist.github.com/cy384/dd638ba8a37c22766444

If you want to use this as a USB keyboard, though, you'll need to get a Leonardo, Pro Micro, or other Arduino with an Atmega32U4.

closed 2016-01-05 by cy384

alchle, 2016-01-06:
Worked great.  Thank you.  I ordered a Leonardo.

Great work!

-Allen

techsetonyoutube, 2017-03-31:
get a Digispark board there small and cheap! 

http://www.ebay.com/itm/like/191544650607?lpid=82&chn=ps&ul_noapp=true