cy384/ppk_usb #9: Unable to upload to Pro Micro

trizex, 2017-07-12:
Hello, I hope you can shed some light on the issue that i'm having. I bought a knockoff from aliexpress

https://www.aliexpress.com/item/New-Pro-Micro-ATmega32U4-5V-16MHz-Replace-ATmega328-for-Ar-Pro-Mini/1859136734.html

It identifies itself in IDE as Arduino Leonardo, but once i try to upload the file, it comes back with a
"KEY_LEFT_GUI" not declared in this scope. Basically the same issue you already had reported by somebody else, so I tried your debugging file (nokeyboard), and....that works. Do you have any ideas as to why I'm experiencing this issue, is the knockoff to blame or? The board looks damn near identical to the one you used. Thank you in advance, hope you can help me out.

trizex, 2017-07-12:
Rhis is the complete error log

Arduino: 1.8.3 (Windows 7), Board: "Arduino Leonardo"

.....ppk_usb-master\ppk_usb-master\arduino_ppk_usb\arduino_ppk_usb.ino: In function 'void config_keymap()':

arduino_ppk_usb:88: error: 'KEY_LEFT_GUI' was not declared in this scope

   key_map[0b00001000] = KEY_LEFT_GUI; // "CMMD" or "Cmd"

                         ^

arduino_ppk_usb:108: error: 'KEY_CAPS_LOCK' was not declared in this scope

   key_map[0b00011000] = KEY_CAPS_LOCK;

                         ^

arduino_ppk_usb:109: error: 'KEY_TAB' was not declared in this scope

   key_map[0b00011001] = KEY_TAB;

                         ^

arduino_ppk_usb:110: error: 'KEY_LEFT_CTRL' was not declared in this scope

   key_map[0b00011010] = KEY_LEFT_CTRL;

                         ^

arduino_ppk_usb:121: error: 'KEY_LEFT_ALT' was not declared in this scope

   key_map[0b00100011] = KEY_LEFT_ALT;

                         ^

arduino_ppk_usb:140: error: 'KEY_BACKSPACE' was not declared in this scope

   key_map[0b00110010] = KEY_BACKSPACE;

                         ^

arduino_ppk_usb:159: error: 'KEY_RETURN' was not declared in this scope

   key_map[0b01000001] = KEY_RETURN;

                         ^

arduino_ppk_usb:169: error: 'KEY_UP_ARROW' was not declared in this scope

   key_map[0b01001001] = KEY_UP_ARROW;

                         ^

arduino_ppk_usb:178: error: 'KEY_DELETE' was not declared in this scope

   key_map[0b01010000] = KEY_DELETE;

                         ^

arduino_ppk_usb:179: error: 'KEY_LEFT_ARROW' was not declared in this scope

   key_map[0b01010001] = KEY_LEFT_ARROW;

                         ^

arduino_ppk_usb:180: error: 'KEY_DOWN_ARROW' was not declared in this scope

   key_map[0b01010010] = KEY_DOWN_ARROW;

                         ^

arduino_ppk_usb:181: error: 'KEY_RIGHT_ARROW' was not declared in this scope

   key_map[0b01010011] = KEY_RIGHT_ARROW;

                         ^

arduino_ppk_usb:188: error: 'KEY_LEFT_SHIFT' was not declared in this scope

   key_map[0b01011000] = KEY_LEFT_SHIFT;

                         ^

arduino_ppk_usb:189: error: 'KEY_RIGHT_SHIFT' was not declared in this scope

   key_map[0b01011001] = KEY_RIGHT_SHIFT;

                         ^

......\ppk_usb-master\ppk_usb-master\arduino_ppk_usb\arduino_ppk_usb.ino: In function 'void config_fnkeymap()':

arduino_ppk_usb:200: error: 'KEY_HOME' was not declared in this scope

   fn_key_map[0b01010001] = KEY_HOME; // left arrow

                            ^

arduino_ppk_usb:201: error: 'KEY_PAGE_DOWN' was not declared in this scope

   fn_key_map[0b01010010] = KEY_PAGE_DOWN; // down arrow

                            ^

arduino_ppk_usb:202: error: 'KEY_END' was not declared in this scope

   fn_key_map[0b01010011] = KEY_END; // right arrow

                            ^

arduino_ppk_usb:203: error: 'KEY_PAGE_UP' was not declared in this scope

   fn_key_map[0b01001001] = KEY_PAGE_UP; // up arrow

                            ^

arduino_ppk_usb:204: error: 'KEY_ESC' was not declared in this scope

   fn_key_map[0b00011001] = KEY_ESC; // tab key

                            ^

arduino_ppk_usb:205: error: 'KEY_F1' was not declared in this scope

   fn_key_map[0b00000000] = KEY_F1; // 1

                            ^

arduino_ppk_usb:206: error: 'KEY_F2' was not declared in this scope

   fn_key_map[0b00000001] = KEY_F2; // 2

                            ^

arduino_ppk_usb:207: error: 'KEY_F3' was not declared in this scope

   fn_key_map[0b00000010] = KEY_F3; // 3

                            ^

arduino_ppk_usb:208: error: 'KEY_F4' was not declared in this scope

   fn_key_map[0b00000100] = KEY_F4; // 4

                            ^

arduino_ppk_usb:209: error: 'KEY_F5' was not declared in this scope

   fn_key_map[0b00000101] = KEY_F5; // 5

                            ^

arduino_ppk_usb:210: error: 'KEY_F6' was not declared in this scope

   fn_key_map[0b00000110] = KEY_F6; // 6

                            ^

arduino_ppk_usb:211: error: 'KEY_F7' was not declared in this scope

   fn_key_map[0b00000111] = KEY_F7; // 7

                            ^

arduino_ppk_usb:212: error: 'KEY_F8' was not declared in this scope

   fn_key_map[0b00110100] = KEY_F8; // 8

                            ^

arduino_ppk_usb:213: error: 'KEY_F9' was not declared in this scope

   fn_key_map[0b00110101] = KEY_F9; // 9

                            ^

arduino_ppk_usb:214: error: 'KEY_F10' was not declared in this scope

   fn_key_map[0b00110110] = KEY_F10; // 0

                            ^

arduino_ppk_usb:215: error: 'KEY_F11' was not declared in this scope

   fn_key_map[0b00110000] = KEY_F11; // -

                            ^

arduino_ppk_usb:216: error: 'KEY_F12' was not declared in this scope

   fn_key_map[0b00110001] = KEY_F12; // =

                            ^

.....\ppk_usb-master\ppk_usb-master\arduino_ppk_usb\arduino_ppk_usb.ino: In function 'void setup()':

arduino_ppk_usb:348: error: 'Keyboard' not found. Does your sketch include the line '#include <Keyboard.h>'?
   Keyboard.begin();

   ^

......\ppk_usb-master\ppk_usb-master\arduino_ppk_usb\arduino_ppk_usb.ino: In function 'void loop()':

arduino_ppk_usb:379: error: 'Keyboard' not found. Does your sketch include the line '#include <Keyboard.h>'?
         Keyboard.releaseAll();

         ^

arduino_ppk_usb:389: error: 'Keyboard' not found. Does your sketch include the line '#include <Keyboard.h>'?
             Keyboard.release(key_code);

             ^

arduino_ppk_usb:393: error: 'Keyboard' not found. Does your sketch include the line '#include <Keyboard.h>'?
             Keyboard.press(key_code);

             ^

exit status 1
'KEY_LEFT_GUI' was not declared in this scope

trizex, 2017-07-12:
So for future reference, I managed to get it working. It was missing a line ,the IDE told me but I wasn't listening :D 

"#include <Keyboard.h>"

And voilla, it wooooorks :D

cy384, 2017-07-12:
glad you got it working!

closed 2017-07-12 by cy384