cy384/ppk_usb #10: using a usb to serial converter?

bijlevel, 2017-10-13:
This is a shot in the dark, but just using a usb to serial converter, like http://www.sigma-shop.com/product/79/modulo-usb-seriale-ft232rl-usb-b-ub232r-atmel-avr-microchip-pic-arduino.html, would that work? Or is it indeed necessary to use a programmable board to get the right key codes? TIA!

cy384, 2017-10-13:
It's probably possible to get data from the keyboard using an RS232 adapter, but you'd also have to do something to turn that into keyboard input on the computer, and it wouldn't be any cheaper or simpler, so I think the arduino-based solution as done in this project is the most reasonable option.

closed 2017-10-13 by cy384

bijlevel, 2017-10-14:
Thanks for the reply! I will try to implement your solution.

Regards,
Sjoerd W. Bijleveld