cy384/ppk_usb #11: Not sure how to s

fionnafire, 2018-03-14:


fionnafire, 2018-03-14:
oof. I have hardly used Github

cy384, 2018-03-15:
thanks for your interest, the files can be opened with [OpenSCAD](http://www.openscad.org/), and it's pretty easy to generate an STL.  As I'm traveling right now, it might be a few days, but I'll upload some as soon as I can.

closed 2018-03-15 by cy384

fionnafire, 2018-03-15:
thank you very much! I can't wait to get this thing working on my phone.

cy384, 2018-05-02:
Sorry for the delayed response, coursework has kept me very busy for the past two months (finals ended yesterday).  [I've added a 1.0.0 release and attached the compiled STLs.](https://github.com/cy384/ppk_usb/releases/tag/v1.0.0)

If you need to tweak the files, you can also try pasting the code into [this in-broswer openscad.](http://openscad.net/)

fionnafire, 2018-05-03:
Thank you so much. You of course had no obligation to make the STL files for me, I really appreciate your work!

I hope your finals went well, and honestly my own college and work schedule would have kept me from being able to make this thing myself anyway. So no worries about the delay.