cy384/ppk_usb #13: Bluetooth connectivity

hmngsghst, 2019-04-26:
Hello! Thanks for sharing this cool mod! Have started to fiddle with electronics, found a PPK at the flea market, googled for an adapter and was happy to find your mod. Am thinking about doing it and was wondering about the possibility of BT connectivity with new BLE-equipped 32u4 boards like the Adafruit Feather 32u4 Bluefruit LE. It seems possible, but without any experience with Arduinos and not much knowledge in electronics and programming, would it be too much of a hassle/hard and too much for a beginner to get this working via bluetooth? Would it need for example power on/off buttons and such? Will be doing it anyway with or without the bluetooth and am ready to put some work into it as a learning experience.

cy384, 2019-04-27:
I'm sure it's possible, though I can't say how hard it would be for a beginner.  I think a person could get it working pretty easily based on the tutorials for the feather/bluefruit boards.  The 3D printed parts won't be sized correctly for the board with a battery, and exact pin numbers used will be different, but those should be relatively straight-forward to figure out.  You could probably just have a slide switch on the battery connection for the power.

hmngsghst, 2019-04-30:
Reading up on the stuff it doesn't seem too hard... Still looking at different possibilities. Might get a bit bulky with the battery, and the cost too..Thanks for the help!