cy384/ppk_usb #15: for Handspring

aoshimak, 2020-08-31:
I tried for Palm3 and Palm5 keyboards as like below. They works well.
https://github.com/milestogo/qmk_firmware/tree/master/keyboards/converter/palm_usb

But I also try to make Handspring one, it is not works well, no keyboard signals goes to PC. 
I use ProMicro with self compiled firmware, wired like this from this information:
Handspring: [VCC/TXD, NC, NC, NC, GND, NC, NC, RXD]

ProMicro      Handspring
2             1(VCC/TXD)
3             8(RXD)
GND           5(GND)
On ProMicro, 10kohm register connects between 3 to GND for pull-up. 

If you have any ideas about Handspring devices, thanks in any advice.

cy384, 2020-09-01:
I think the only important difference with the Handspring devices is that the handshake bytes it sends will be "F9 FB" instead of "FA FD", so you could try changing that in the code, looks like the function "palm_handshake" on the link you gave.

hope that helps!

aoshimak, 2020-09-02:
Thank you for your reply. 
On qmk palm_usb converter, that is based on your code.
And Handspring option seems on config.h

Then, who is the transporter from your code to qmk?

cy384, 2020-09-02:
I don't know, sorry.

FedX-sudo, 2021-10-03:
I just got myself a stowaway keyboard thinking this would, work. Turns out its the Handspring version, and I was wondering if there was any progress getting the Hanspring working. Thanks!

clach04, 2025-02-20:
> Thank you for your reply. On qmk palm_usb converter, that is based on your code. And Handspring option seems on config.h
> 
> Then, who is the transporter from your code to qmk? Please tell me if you know.

I **think** https://github.com/milestogo/qmk_firmware/tree/palm_usb