cy384/ppk_usb #16: Keyboard stuck most of the time on "waiting for keyboard serial ID..."

VectorKappa, 2020-09-13:
Hello. I've been using a german version of palm keyboard (V version) for a while now, and some time ago the keyboard stopped getting recognized on my android phone. Sometimes replugging the adapter worked, but lately it doesn't work at all.

I tried placing something in the while loop, to prevent my phone from going to low-power otg mode, but to no avail.
Very rarely, recompiling fixes the issue, but only for one launch.

As far as I can see, everything seems to be soldered on correctly, but could it be the cat.5 wire fault? It looks flimsy. I could switch it to some silver-plated copper wire, but I don't want to potentially waste it.

cy384, 2020-09-13:
I think the most likely issue is the contact between the wire on the adapter and the connector on the keyboard.  You can try to poke them around to see if that improves the connection.

I've been considering redesigning the 3D printed part to hold the wires better, the current design isn't perfect.

VectorKappa, 2020-09-14:
Alright, that was entirely my fault for using a cheap cat5 found in my toolbox. I scraped the patina off the wires with a small file, works like a charm. I'll probably replace the wires.

Thanks for the response!

closed 2020-09-14 by VectorKappa