cy384/ppk_usb #17: HP Pocket PC version?

joeskeen, 2020-12-31:
I used to have one of these keyboards for my old palm pilot, but somehow it got lost. I just picked up one of these stowaway keyboards, but it was built for Pocket PC, Windows key and all. I'm planning to convert mine into a general purpose usb-c keyboard, but I found an image on the web that makes me wonder if I could do the conversion without an Arduino:

https://newatlas.com/thanko-full-size-usb-folding-keyboard/14897/

This is definitely the same keyboard, but if you take a look at the pictures, one shows a usb cable attached. I'm wondering if I remove that panel, perhaps there is a ready-for-usb circuit I can attach to? If so, this could greatly simplify the conversion process for all stowaway keyboards.

Wish me luck!

joeskeen, 2020-12-31:
Ok, so I got it apart, and didn't find some magical USB connection spot on the PCB (hey, I can dream, can't I?)

[image]

My guess is that the picture I found featured a different PCB under the hood.

joeskeen, 2020-12-31:
After doing a little more research, I think I'll use this project with an Arduino Seeeduino XIAO - the size of a quarter and costs under $5. The plan is I can replace all the wiring on the left for the pocket PC with the board and 3D print an enclosure for it.

https://www.seeedstudio.com/Seeeduino-XIAO-Arduino-Microcontroller-SAMD21-Cortex-M0+-p-4426.html

cy384, 2020-12-31:
some changes will need to be made to the code, but it seems like all of the details for pocketpc are available in the reference document I link in the readme, and it's nothing particularly hard.

a problem you may have with that board is that it only supports 3.3V, not 5V, and the keyboard electronics probably need to run and interface at 5V.

joeskeen, 2020-12-31:
That's some good information, thanks!

cy384, 2021-01-06:
now that I look at the PCB again, I think it has a voltage regulator on it, it may run fine at 3.3V if you bypass that, when I have a chance I'll look at the voltages on mine